stack-of-tasks/pinocchio #16: se3 <-> XYZQUAT conversion

aelkhour, 2015-04-21:
The se3 <-> XYZQUAT conversion functions in src/python/utils.py seem to be bugged, but I'm not sure where.

If I run the following script:
```python
from pinocchio.utils import *

xyzquat = [0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
print xyzquat
M = XYZQUATToSe3(xyzquat)
xyzquat_2 = se3ToXYZQUAT(M)
print xyzquat_2
```

I get:
```
[0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
[0.0, 0.0, 1.0, 1.0, 0.0, 0.0, 0.0]
```

aelkhour, 2015-04-21:
Actually I think I found the error:

From eigenpy/src/quaternion.hpp, it seems the [] operator does not have the same order as the arguments in constructor:
```c++
	.def(bp::init<Scalar,Scalar,Scalar,Scalar>
	     ((bp::arg("w"),bp::arg("x"),bp::arg("y"),bp::arg("z")),
	      "Initialize from coefficients.\n\n"
	      "... note:: The order of coefficients is *w*, *x*, *y*, *z*. "
	      "The [] operator numbers them differently, 0...4 for *x* *y* *z* *w*!"))
```

Therefore utils.se3ToXYZQUAT should be written as:
```python
def se3ToXYZQUAT(M):
    xyz = M.translation
    quat = se3.Quaternion(M.rotation).coeffs()
    config = [  float(xyz[0,0]), float(xyz[1,0]), float(xyz[2,0]),
                float(quat[0,0]), float(quat[1,0]), float(quat[2,0]), float(quat[3,0]) ]
    return config
```

fvalenza, 2015-04-21:
Hello Antonio and thanks for this feedback.
I am taking a look and the problem is the use of different conventions at the same time:
1/ Eigen uses (w, x, y, z) to construct a quaternion
      BUT return it as (x, y, z, w) by the method ::coeffs()
2/ In Pinocchio, we use the convention (x, y, z, w) .
Therefore, XYZQUATToSe3 seems ok for me.
The solution is to change se3ToXYZQUAT:

def se3ToXYZQUAT(M):
    xyz = M.translation
    quat = se3.Quaternion(M.rotation).coeffs()
    config = [  float(xyz[0,0]), float(xyz[1,0]), float(xyz[2,0]),
                float(quat[0,0]), float(quat[1,0]), float(quat[2,0]), float(quat[3,0]) ]
    return config

Unfortunately, this will bring a little "bug" with the Gepetto viewer. ( (w, x, y, z) convention).

In conclusion the issue is waiting for a correction that will be done by the end of the week (24/04) normally and will come with a new release.
Let me know if it is ok for you.

Note: issue will be closed when fixed

aelkhour, 2015-04-21:
@fvalenza Thanks, fine with me :-)

fvalenza, 2015-05-19:
The version 1.0.1 has been released with fix. Closing pull request

closed 2015-05-19 by fvalenza