stack-of-tasks/pinocchio #19: [Python] Overwritting of robot.q0

fvalenza, 2015-04-28:
I am testing python scripts (to record for a video) and I have a very weird behavior.
I don't remember having this problem before but even when i come back to v1.0.0, I have this same issue.
Is it happening just to me ? ( may be due to my computer) or are you able to reproduce this bug ?

```python
import numpy as np
from pinocchio.romeo_wrapper import RomeoWrapper

robot = RomeoWrapper("/home/fvalenza/local/devel/pinocchio/models/romeo.urdf")
print robot.q0

q=robot.q0
x_rw_des = 3
k = 0.05

for i in range(100):
    Mrw = robot.position(q,32)
    e = Mrw.translation.item(0) - x_rw_des
    J = Mrw.rotation * robot.Jrh(q)[:3,:]
    J = J[0,:]
    qdot = -J.T*e
    qdot = qdot*k
    robot.increment(q,qdot)

print robot.q0
```

with outputs:

```
[[ 0.         0.         0.840252   0.         0.         0.         1.         0.
   0.        -0.3490658  0.6981317 -0.3490658  0.         0.         0.
  -0.3490658  0.6981317 -0.3490658  0.         0.         1.5        0.6
  -0.5       -1.05      -0.4       -0.3       -0.2        0.         0.         0.
   0.         1.5       -0.6        0.5        1.05      -0.4       -0.3
  -0.2      ]]
```
```
[[ 2.53137626  0.01812421  0.84399951  0.00609353 -0.01829379  0.20671735
   0.97821069  0.          0.         -0.3490658   0.6981317  -0.3490658
   0.          0.          0.         -0.3490658   0.6981317  -0.3490658
   0.          0.41576846  1.5         0.6        -0.5        -1.05       -0.4
  -0.3        -0.2         0.          0.          0.          0.
   1.16517659 -0.97272307  0.72214912  0.79606203 -0.4        -0.3        -0.2       ]]
```

aelkhour, 2015-04-28:
It seems you're incrementing robot.q0. It's just a hunch, but you might need to copy robot.q0 into q instead of just referencing it.
See http://docs.scipy.org/doc/numpy/reference/generated/numpy.copy.html

fvalenza, 2015-04-28:
Problem solved. Was due to my misunderstanding of python
Thank you @aelkhour

closed 2015-04-28 by fvalenza