stack-of-tasks/pinocchio #27: Parse fixed joints from URDF

aelkhour, 2015-06-15:
Some robot URDF files contain semantic information (wrist, ankle, sole, etc.) which is stored as a pair (fixed joint, empty link). For now, it seems such joints are not parsed in the robot structure, and thus cannot be used in control and/or planning.
Is there any computational limitation that would prevent storing the fixed joints, and computing their positions and jacobians whenever needed?

fvalenza, 2015-07-27:
Currently fixed joints are handled in the way that the following link  parameters (inertia, mass..) are aggregated to its parent and/or repercuted to its children ( for relative position for exemple). But there is no entity called "fixed joint" for which we could perform these computations and adding this kind of joint as it in the kinematic chain may need some tricks i think

A solution would be to add a list a spot in the model (storing the parent joint, relative position to its parent, and its name) and then compute positions and jacobians when needed.

If it is ok for you, we may add it to future releases

aelkhour, 2015-07-28:
This sounds like a good solution, thanks.

olivier-stasse, 2015-12-17:
Hi,
Just a quick feedback on this point. I recently parsed a URDF model (tiago) with the devel branch of pinocchio. The URDF model contains tons of fixed frame and its works beautifully for my application.
So thanks a lot ! P.S. : I did not try the Jacobian computation.

jcarpent, 2016-02-10:
#96 solves the issue. Closing.

closed 2016-02-10 by jcarpent