stack-of-tasks/pinocchio #33: Matlab interface

aelkhour, 2015-07-24:
A Matlab interface similar to the Python interface could be useful to Matlab developers.
Potential users:
  - Control engineers who develop controllers in Matlab
  - Mechanical engineers who have a limited programming experience besides Matlab

aelkhour, 2015-10-16:
I have a question related to this issue:

There seems to be an ongoing effort to provide [Matlab support in swig](https://github.com/jaeandersson/swig). Did you give any thought to using swig for the python interface generation?

nmansard, 2015-10-16:
Hello, 
Yes indeed. I lightly benchmarked several python interface systems in terms of ease of use and efficiency. Boost seemed to be the best option. 
The binding of eigen and mummy is indeed quite tricky. I would be a priori very reluctant to spend time doing it a second time. 
On the other hand, matlab interfaces are very light to implement. Do you really thing that swing would help, if not for capitalising on the python bindings? 
N

aelkhour, 2015-10-19:
A couple of questions:
- What is mummy?
- Is Eigen memory alignment the only tricky part or did you encounter other problems?

I am neither familiar with matlab interfaces nor with swig (at least not yet), and I am quite happy with the current state of the python bindings. So I would give swig a try only if it would make writing matlab interfaces faster than doing a raw binding (through mexFunction I guess).

jcarpent, 2015-10-19:
One quick answer: mummy is in fact numpy.

nmansard, 2015-10-19:
Mummy is indeed the best my phone could propose to correct Numpy, as already corrected by Justin. Sorry for the typo. 

The tricky part in boost::python was the memory alignment (or more exactly, the memory allocation of Eigen memory by boost::python, that does not respect Eigen alignment). The tricky part in general is to bind Eigen data in C with Numpy data in Python. It would be more straightforward to wrap Eigen data in Python. Wrapping is for example what is done for Pinocchio::Model or Pinocchio::Force classes: you are not using a Python object to make the equivalence with the C object, but directly manipulating the C object in Python. 

I think binding Eigen with Numpy provides better functionalities in the end, however is more difficult to implement. For what I benchmarked initially, it seems equality tricky with the various Python wrapping/binding frameworks. 

Again in my opinion, wrapping is much easier in Matlab, because Matlab is not strongly typed. Programming the matlab bindings could be done with much effort with the native MEX system of Matlab.

aelkhour, 2015-10-19:
Thanks for the clarification, I will give the matlab MEX a try.

jcarpent, 2016-01-16:
Have you implemented any Matlab interface? If not, are you still interested by such feature?

aelkhour, 2016-01-18:
Sadly I haven't had the time to implement anything, but we are still clearly interested in such a feature.

cmastalli, 2018-11-23:
I would like to add a comment to this discussion line. Indeed it's very tricky to bind Eigen data with Numpy. During the period at IIT, I used SWIG to bind Eigen data. Basically you need to write an interface file, this file can be re-used for any project. Here you find the file: 
https://github.com/robot-locomotion/dwl/blob/master/swig/eigen.i

closed 2021-08-24 by jcarpent

traversaro, 2021-08-25:
If anyone ever needs that, a possible strategy to test is to use the Pinocchio Python bindings in MATLAB. Recent version of MATLAB have quite advanced Python interoperability features, see https://www.mathworks.com/help/matlab/call-python-libraries.html, https://github.com/dic-iit/bipedal-locomotion-framework/issues/53#issuecomment-789058776 and https://github.com/robotology/robotology-superbuild/issues/760 .

jcarpent, 2021-08-25:
Fantastic. I didn't know that.
We may give it a try once. @proyan Would you may give it a try?

@traversaro What is your feedback on it?

traversaro, 2021-08-25:
> @traversaro What is your feedback on it?

We only tried it for some examples for now (see https://github.com/dic-iit/bipedal-locomotion-framework/issues/53#issuecomment-789058776) but it seems quite promising, but to be honest we never had any serious use of it (like, a student that did all the computation for a paper/thesis in it). To quickly test it, if you have access a possible option is to install pinocchio via conda in [MATLAB Online](https://matlab.mathworks.com/). We have a simple script in https://github.com/robotology/robotology-superbuild/blob/0f82a33dca1fe46111945e09382df26e39c0efb5/doc/matlab-one-line-install.md#installation-on-matlab-online for installing conda packages in MATLAB Online with a few commands. 
To use the script with Python Pinocchio bindings, probably it would be sufficient to:
* the script should be modified around https://github.com/robotology/robotology-superbuild/blob/d94d8ca32e5b1e446ad9ec835565e315cceb7bac/scripts/install_robotology_packages.m#L68 to  install pinocchio via conda forge instead of the robotology packages
* the selection of the python interpreter of conda should be added to the activation script around https://github.com/robotology/robotology-superbuild/blob/d94d8ca32e5b1e446ad9ec835565e315cceb7bac/scripts/install_robotology_packages.m#L72 , adding a line such as `pyversion(fullfile(robotology_install_prefix,'/bin/python3'))` or something similar.