stack-of-tasks/pinocchio #45: CRBA produces wrong results

martinfelis, 2015-09-30:
The joint-space inertia matrix computed using pinocchio.crba is inconsistent with pinocchio.rnea. 
To check whether the Composite Rigid Body Algorithm produces the correct results one can leverage the Recursive Newton-Euler Algorithm to compute the joint-space inertia matrix column by column.

I created the following gist:

https://gist.github.com/martinfelis/ea8a9fc30d24d2968274

which computes the inertia matrix both using crba and multiple calls to rnea.

When calling with the simple_humanoid.urdf model I get the following output:
```
Model: simple_humanoid.urdf
--
Stats      |  #q | #qdot 
 Pinocchio |  29 |  29
--
Error M[0:6,0:6]
[[  7.85e-001  -1.94e-003  -1.67e-002   0.00e+000   0.00e+000   2.22e-016]
 [ -1.94e-003  -4.49e-001  -2.78e-017  -4.32e-002   8.88e-016   0.00e+000]
 [ -1.67e-002  -2.78e-017   0.00e+000   0.00e+000   1.15e-321   1.80e-016]
 [  0.00e+000  -4.32e-002   0.00e+000  -4.32e-002   8.88e-016   0.00e+000]
 [  0.00e+000   4.44e-016   1.15e-321   0.00e+000   8.88e-016   0.00e+000]
 [  0.00e+000   0.00e+000  -4.16e-017   0.00e+000   0.00e+000   0.00e+000]]
Error M[0:6,6:12]
[[ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]]
```
Note the errors in the first two diagonal elements with an order of magnitude of 10^-1. Possibly off-diagonal elements are okay.

However when running with the ```-f``` flag to use a free-flyer joint I get 

```
Model: simple_humanoid.urdf
--
Stats      |  #q | #qdot 
 Pinocchio |  36 |  35
--
Error M[0:6,0:6]
[[  0.00e+00   0.00e+00   0.00e+00  -0.00e+00   1.24e-14  -0.00e+00]
 [  0.00e+00   0.00e+00   0.00e+00   1.78e-15  -0.00e+00  -1.42e-14]
 [  0.00e+00   0.00e+00  -1.71e-13   0.00e+00  -2.13e-13  -0.00e+00]
 [ -0.00e+00   1.78e-15   1.42e-14   3.83e+00  -1.39e-17  -3.22e-01]
 [ -5.33e-15  -0.00e+00  -1.07e-14  -3.57e-15  -3.83e+00  -7.33e-15]
 [ -3.55e-15   0.00e+00  -0.00e+00  -3.22e-01  -3.77e-15  -7.11e-15]]
Error M[0:6,6:12]
[[  0.00e+00   1.24e-14   6.38e-16   4.44e-15   2.50e-15   0.00e+00]
 [  0.00e+00   0.00e+00  -3.52e-15   0.00e+00   0.00e+00  -2.55e-15]
 [ -1.78e-13  -7.46e-14   0.00e+00  -7.46e-14  -7.46e-14   0.00e+00]
 [  7.85e-01  -1.94e-03  -1.67e-02  -5.58e-15  -5.58e-15  -3.33e-15]
 [ -1.73e-03  -4.49e-01  -2.69e-15  -4.32e-02  -4.88e-15   0.00e+00]
 [ -3.14e-02   9.40e-02   1.78e-15   1.22e-15   1.55e-15  -7.08e-16]]
```
Where also in the off-diagonal blocks are large errors.

jcarpent, 2015-09-30:
Hello Martin,

Thanks for your report. 
I tried your code on my computer (Mac OS X 10.10, llvm 3.4, Eigen 3.2.5, last commit of Pinocchio on the devel branch 227a4cde) and everything looks fine with very small residues:

```
Model: ../pin-fix-bug/models/simple_humanoid.urdf
--
Stats      |  #q | #qdot 
 Pinocchio |  36 |  35
--
Error M[0:6,0:6]
[[  0.00e+00   0.00e+00   0.00e+00  -0.00e+00   1.24e-14  -0.00e+00]
 [  0.00e+00   0.00e+00   0.00e+00   1.78e-15  -0.00e+00  -1.42e-14]
 [  0.00e+00   0.00e+00  -1.71e-13   0.00e+00  -2.13e-13  -0.00e+00]
 [ -0.00e+00   1.78e-15   1.42e-14   1.42e-14   0.00e+00   4.44e-16]
 [ -5.33e-15  -0.00e+00  -1.07e-14  -3.55e-15  -1.42e-14  -7.33e-15]
 [ -3.55e-15   0.00e+00  -0.00e+00   4.44e-16  -3.77e-15  -7.11e-15]]
Error M[0:6,6:12]
[[  0.00e+00   1.24e-14   6.38e-16   4.44e-15   2.50e-15   0.00e+00]
 [  0.00e+00   0.00e+00  -3.52e-15   0.00e+00   0.00e+00  -2.55e-15]
 [ -1.78e-13  -7.46e-14   0.00e+00  -7.46e-14  -7.46e-14   0.00e+00]
 [  1.78e-15  -5.58e-15  -4.41e-15  -5.58e-15  -5.58e-15  -3.33e-15]
 [  4.34e-19  -7.11e-15  -2.69e-15  -5.33e-15  -4.88e-15   0.00e+00]
 [  2.66e-15  -2.11e-15   1.78e-15   1.22e-15   1.55e-15  -7.08e-16]]
```

Can you give me your current configuration in order to reproduce your errors ?
Best,

fvalenza, 2015-09-30:
Hello Martin,
Here are some results on an Ubuntu 12.04 64 bits, with gcc 4.6.3, eigen 3.2.1
```
Model: /local/fvalenza/devel/src/pinocchio/models/simple_humanoid.urdf
--
Stats      |  #q | #qdot 
 Pinocchio |  29 |  29
--
Error M[0:6,0:6]
[[  0.00e+00   0.00e+00   4.72e-16   0.00e+00   0.00e+00   2.22e-16]
 [  0.00e+00   0.00e+00  -2.78e-17   0.00e+00   8.88e-16   0.00e+00]
 [  2.78e-17  -2.78e-17   0.00e+00   0.00e+00   0.00e+00   1.80e-16]
 [  0.00e+00   1.78e-15   0.00e+00   0.00e+00   8.88e-16   0.00e+00]
 [  0.00e+00   4.44e-16   0.00e+00   0.00e+00   8.88e-16   0.00e+00]
 [  0.00e+00   0.00e+00  -4.16e-17   0.00e+00   0.00e+00   0.00e+00]]
Error M[0:6,6:12]
[[ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.]]

```
With the FreeFlyer :
```
Model: /local/fvalenza/devel/src/pinocchio/models/simple_humanoid.urdf
--
Stats      |  #q | #qdot 
 Pinocchio |  36 |  35
--
Error M[0:6,0:6]
[[  0.00e+00   0.00e+00   0.00e+00  -0.00e+00   1.24e-14  -0.00e+00]
 [  0.00e+00   0.00e+00   0.00e+00   1.78e-15  -0.00e+00  -1.42e-14]
 [  0.00e+00   0.00e+00  -1.71e-13   0.00e+00  -2.13e-13  -0.00e+00]
 [ -0.00e+00   1.78e-15   1.42e-14   1.42e-14   0.00e+00   4.44e-16]
 [ -5.33e-15  -0.00e+00  -1.07e-14  -3.55e-15  -1.42e-14  -7.33e-15]
 [ -3.55e-15   0.00e+00  -0.00e+00   4.44e-16  -3.77e-15  -7.11e-15]]
Error M[0:6,6:12]
[[  0.00e+00   1.24e-14   6.38e-16   4.44e-15   2.50e-15   0.00e+00]
 [  0.00e+00   0.00e+00  -3.52e-15   0.00e+00   0.00e+00  -2.55e-15]
 [ -1.78e-13  -7.46e-14   0.00e+00  -7.46e-14  -7.46e-14   0.00e+00]
 [  1.78e-15  -5.58e-15  -4.41e-15  -5.58e-15  -5.58e-15  -3.33e-15]
 [  4.34e-19  -7.11e-15  -2.69e-15  -5.33e-15  -4.88e-15   0.00e+00]
 [  2.66e-15  -2.11e-15   1.78e-15   1.22e-15   1.55e-15  -7.08e-16]]
```

I get the same results as Justin when using clang

aelkhour, 2015-09-30:
```c++
// Y = R' L   ===> (12 m + 8 a)
const Matrix2 & Y = R.template block<2,3>(1,0) * L;
// const Matrix2 Y(R.template block<2,3>(1,0) * L);

jcarpent, 2015-09-30:
You want to mention that
```
const Matrix2 Y(R.template block<2,3>(1,0) * L);
```
gives the good result ?

martinfelis, 2015-09-30:
Hrmm.... I am on:
* Ubuntu 14.04
* g++ (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
* eigenpy 5b9029e7d59
* libeigen3-dev 3.2.0-8

Both master and devel branch give the same results. The culprit seems to be using Release mode. In Debug mode the errors are negligible.

jcarpent, 2015-09-30:
It seems that this error already occurred for @aelkhour but was never fixed as @aelkhour switched to clang as far as I remember. 
Does the line 
```c++
const Matrix2 Y(R.template block<2,3>(1,0) * L);
```
fixe your problem (@martinfelis) ?

martinfelis, 2015-09-30:
It does indeed.

jcarpent, 2015-09-30:
Thanks Martin for reporting and checking. Thanks Antonio for highlighting the buggy line. The bug will be fixed quite soon.

closed 2015-09-30 by jcarpent

aelkhour, 2015-09-30:
Thanks for the hotfix @jcarpent and @fvalenza.