stack-of-tasks/pinocchio #47: Enforcement of One Definiton Rule

aelkhour, 2015-10-05:
While trying to use the C++ API, I encoutered several build errors related to multiple definitions.
This happens mainly:
- in the operators of the joint classes. As pinocchio is header only, all templated functions must be inline to avoid multiple definitions. I can make a PR for that if you agree.
- in model.hpp, with the definition of static const member se3::model::gravity981 in the header file. One solution would be to move the definiton to a .cpp file, forcing the client code to link against libpinocchio. Another solution would be to remove the definition and force the client code to define it when needed, which is not very nice. Other propositions are more than welcome.

jcarpent, 2015-10-05:
Concerning the model.hpp, it is an ongoing work and it will be released quite soon. The solution of the lib seems to be the best.
For the joint classes, it may really help us if you inline all the functions via a PR.

Thanks for reporting.

aelkhour, 2015-10-07:
Alright I'll make a PR adding the missing inlines.

jcarpent, 2015-10-17:
It seems that that this bug has been fixed recently with PR #54 and #49. Thanks to @aelkhour for this fixing.

aelkhour, 2015-10-17:
I think the issue should be kept opened until se3::model::gravity981 is moved is defined in a .cpp file.

jcarpent, 2015-10-19:
The gravity vector is now part of the pinocchio library (#56)

closed 2015-10-19 by jcarpent