stack-of-tasks/pinocchio #48: Travis build "not passing" randomly

fvalenza, 2015-10-06:
Since Pull Request on dynamic dense joints, travis has troubles to build AND check successfully. 

In fact there are two major problems that occur alternatively:
1/ When compiling the unittest urdf, there is an error in the flags ( there is a "urdf_CFLAGS-NOTFOUND " instead of "-I/usr/include/eigen3 -I/opt/openrobots/include" in the travis log).
I do not understand where this comes from* given that:
  - there are no changes in the travis build configuration
  - there are (almost) no changes in the CmakeLists : just one file added at a certain commit
  - urdfdom is found when generating the Makefile

2/ When error #1 does not occur, travis seem to "freeze" at the end of the build and then abort the procedure. The typical error message at the end of log is:
"  No output has been received in the last 10 minutes, this potentially indicates a stalled build or something wrong with the build itself.
The build has been terminated "
If you go on https://travis-ci.org/stack-of-tasks/pinocchio/jobs/83854014 , you will see that it occured after all the test passed.

Note that it can happens only on gcc build, or clang build, or both.

* I don't know why the flags are -NOTFOUND but i suppose it happens in the unittest/CMakeLists.txt at line 35
GET_TARGET_PROPERTY(${target}_CFLAGS ${target} COMPILE_FLAGS) 
SET_PROPERTY(TARGET ${target} PROPERTY COMPILE_FLAGS "${${target}_CFLAGS} ${flag}")

fvalenza, 2015-10-14:
Pull Request #51 should - partially* - solve error 2/
Commit #c8ff00 solve error 1/

* The random failures rate seems to decrease

jcarpent, 2015-10-17:
It seems that this bug has been fixed via recent PR.

closed 2015-10-17 by jcarpent