stack-of-tasks/pinocchio #69: Floating joint in URDF file

olivier-stasse, 2015-12-17:
Hi,
Why the urdf parser is not able to handle floating joint ?

They are inside the pinocchio package.
From http://wiki.ros.org/urdf/XML/joint it looks like this is still valid from the urdf specification viewpoint.

Is it because the case never occurred or is there a rational behind it ?

Best,
Olivier.

jcarpent, 2015-12-17:
Hi,

With Nicolas, we think that the URDF file must only describe the kinematics chain and not its link to the environment. The idea is then to precise the root joint via the buildModel function specifying the joint type we want.

Do you need the other feature ?

florent-lamiraux, 2015-12-17:
Some urdf files come with the degrees of freedom corresponding to the floating base.  It is necessary to handle those cases by specifying pinocchio not to add a floating joint.

olivier-stasse, 2015-12-17:
Like for the fixed joint, it is something inside the URDF specification. I do not believe that asserting without any option is a nice behavior. It is even more bizarre when the joint freeflyer is inside the library. I am fine if you want this to be the default behavior but I do believe that implementing the other policy and make it optional is possible without a strong impact on the code itself.

closed 2015-12-17 by jcarpent