stack-of-tasks/pinocchio #83: Release process

aelkhour, 2016-02-01:
It has been some time since a new version pinocchio has been released. As a result most of us rely on the devel branch, where then API is broken from time to time (and this is normal on dev branch).

I don't know what the release planning is, but it would be nice to provide "stable" versions a bit more frequently, especially when new features are provided or API is broken.

jcarpent, 2016-02-01:
I fully agree with your remark. The next release will be available soon from the current version of the devel branch.
In the future, we will do our best to launch release more frequently.

Thanks !

olivier-stasse, 2016-02-03:
Dear all, 
I need to include pinocchio in robotpkg and this release issue is very pressing. Thus what should be the next release number. From what I understand from the changes it should be at least 1.1.0, even maybe 2.0.0 ?

jcarpent, 2016-02-03:
Currently, we do not need to go to 2.0.0 because the major modifications do not occur yet. I would go on 1.1.0.

jcarpent, 2016-02-04:
A new release 1.1.0 is now available. Closing issue.

closed 2016-02-04 by jcarpent