stack-of-tasks/pinocchio #89: Coverage - Needs of more unit tests

olivier-stasse, 2016-02-03:
Dear all,
To have a better view of the code state so that we can improve the release process,
I made the link to coveralls. I apologize for the rough process. 
According https://coveralls.io/builds/4952648 they are three files which are not tested at all:
joint_planar.hpp joint_dense.hpp constraint.hpp.
It would be nice to have unit test related to those files.

jcarpent, 2016-02-03:
Thanks Olivier for this great job!
Concerning those files, they are currently not used at least for joint_dense.hpp and constraint.hpp.
I will do my best to integrate some unit test around joint_planar.hpp asap.

jcarpent, 2016-02-03:
And also, all the python bindings are not tested at all. But it won't appear in the coverage test, because they are not compiled on travis.

closed 2016-05-20 by olivier-stasse