stack-of-tasks/pinocchio #93: Parser urdf

florent-lamiraux, 2016-02-04:
In its current version, the urdf parser is not able to understand url starting by "package://"
Moreover, it seems that srdf files are not parsed.

aelkhour, 2016-02-04:
@florent-lamiraux Do you think there's a solution for the `package://` issue without adding a dependency to ROS?

+1 for srdf.

+1 as well for [sdf](http://sdformat.org/spec), the official format for Gazebo simulator. It offers slightly more options than urdf, such as adding parallel spring-dampers on the joints. Although outside the current scope of pinocchio, the model does not need to be a kinematics tree, it can also be a DAG for parallel robots.

florent-lamiraux, 2016-02-04:
I think the best solution is to parse ROS_PACKAGE_PATH environment variable in C++ and to search in the corresponding directories for the files. The way ROS proceeds is not very clear and fails sometimes to find new directories because the right setup.sh has not been resourced after installation.

jcarpent, 2016-02-10:
The final idea is then:
- Add method to directly read a SRDF file to activate collision pairs.
- Add function to locate model and its data (URDF, SRDF, meshes, etc) by only the robot name and seek in $ROS_PACKAGE_PATH 

Have you other requests?

jcarpent, 2016-03-14:
Solved by #110.

closed 2016-03-14 by jcarpent