stack-of-tasks/pinocchio #104: Add integrate and differentiate methods in JointModel

jcarpent, 2016-02-16:


jcarpent, 2016-03-03:
Add also random method for each joint

jmirabel, 2016-03-03:
For the `differentiate` method of SO3 joints, a convention must be chosen. This convention relates to what is being modelled: the rotation group or the unit quaternions group. Each rotation is mapped to 2 quaternions.
Thus, the "difference" between two quaternions `a` and `b` can be:
- either the angle axis between `a` and `b` (the quaternion representation)
- or the angle axis that has the shortest angle, between `a` and `b` or between `a` and `-b` (the rotation representation)

In HPP, we use the rotation representation. This has several implications. The one I can think of now are:
-  Equality check: `a` and `-a` must be considered as the same rotation for the whole software => two configurations `q1 and `q2` can be equal even if  `q1.isApprox (q2)` is false.
- Difference and distance: use the angle axis that has the shortest angle

See https://github.com/jmirabel/hpp-model/blob/devel/src/joint-configuration.cc#L243

jcarpent, 2016-03-04:
The second proposition is the rational one. Because, we are working on base orientation, which can be handled by many representations:
- rotation matrix (unique representation)
- quaternion, with a non-infectivity (a and -a leads to the same orientation)
- Euler angles (with a lot of drawbacks - singularities)

fvalenza, 2016-04-01:
Solved with #141

closed 2016-04-01 by fvalenza