stack-of-tasks/pinocchio #108: Robot Wrapper

jcarpent, 2016-02-22:
Create a robot wrapper similar to the existing python class containing a model and several data store in a vector achieving the interface between (model,data) and the main algorithms.

@aelkhour Do you have some precise requests?

aelkhour, 2016-02-22:
- Not throwing exceptions would be great.
- Not making explicit dynamic allocations (using `new` operator) would be great.

If this conflicts with your development roadmap, then no worries.

jcarpent, 2016-02-22:
Concerning the exceptions, why this precise request?

aelkhour, 2016-02-22:
Our whole code base is supposed to be exception-free. I know this is not currently the case in pinocchio, which is why this is not a hard requirement.

jcarpent, 2016-02-22:
Ok. If you want you can just use the instruction `-fno-exceptions` with gcc.
I think for lambda user, exceptions are mandatory that why we try to populate our code with them.

aelkhour, 2016-02-22:
Noted, thanks.

nmansard, 2016-02-22:
It indeed is: please no throw in construction and destruction (just 
image what can of torture you are generating on the poor stack when 
creating memory with a destruction phase to throw something).

jcarpent, 2018-02-21:
A robot wrapper have been implemented in https://github.com/stack-of-tasks/tsid

closed 2018-02-21 by jcarpent