stack-of-tasks/pinocchio #109: Trying to install pinocchio in Ubuntu 14.04

mjm522, 2016-02-22:
There was no issues while installation (Using installation instruction given). When I try to open it via python in command prompt while trying out the initial examples I get the following error.

 from pinocchio.utils import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named pinocchio.utils

I guess it is not able find the path. But my sys.path has following registered:

echo $PYTHONPATH
/home/ad/pinocchio/src/python:/usr/local/lib/python2.7/site-packages:/home/ad/pinocchio/install/lib64:/home/ad/pinocchio/install/lib/python2.7/site-packages

Could any body please tell me what I have done wrong?

jcarpent, 2016-02-22:
Did you install eigenpy (https://github.com/stack-of-tasks/eigenpy), the Eigen-Numpy biding which is at the root of the python binding of Pinocchio.

For further installation information, you can also visit the following page: https://github.com/stack-of-tasks/pinocchio/wiki/Installation

mjm522, 2016-02-22:
thanks for the reply. When I follow those instructions, i get following error.
W: Failed to fetch http://robotpkg.openrobots.org/packages/debian/dists/trusty/Release  Unable to find expected entry 'robotpkg/binary-i386/Packages' in Release file (Wrong sources.list entry or malformed file)

jcarpent, 2016-02-22:
Indeed, it looks weird. Can you check with `uname -a` in your terminal to check your architecture.

Anyway, you can still install the python bindings, by first installing eigenpy from source and then, updating your PKG_CONFIG_PATH to point to your-install-path/lib/pkgconfig . And then retry the installation of Pinocchio (in fact, it will find the eigenpy dependency with pkg-config and then compile the python lib).

mjm522, 2016-02-24:
yeah Checked it, my architecture is 64bit!

I was able to install the pinocchio with the following modification to the environment variables apart to the one mentioned by jcarpent in above comment. I have made git clone in the home folder and installed it in local.

export PYTHONPATH=/usr/local/lib/python2.7/site-packages/:$PYTHONPATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH

Hope it helps somebody.

jcarpent, 2016-02-24:
Thanks for this reply and your useful feedback. I'll close the issue now.

closed 2016-02-24 by jcarpent