stack-of-tasks/pinocchio #111: setting joint limits

proyan, 2016-03-01:
Hello all,
It is currently difficult to write to Joint limits. I suppose the only way available is if I change the Eigen::VectorXd elements available in Model class (lowerPositionLimit, upperPositionLimit ...), though it doesn't feel clean to do that

Is it possible to get a writable expression in these? Currently the visitor in joint-visitor.hpp returns only a readable value.

jcarpent, 2016-03-01:
You can access to this quantities in data via lowerPositionLimit and upperPositionLimit. They are both in read and write mode.

proyan, 2016-03-01:
Thanks. I was mainly worried that changing limits might require some other changes as well.

jcarpent, 2016-03-01:
Your welcome. I will close the issue.

closed 2016-03-01 by jcarpent