stack-of-tasks/pinocchio #115: Model joint limits

jcarpent, 2016-03-02:
Currently, joint limits are implemented as an algorithm and we consider that joint limits may depend on the current configuration. 
But this option is not satisfying. 
One option is to put joint limit quantities in model and to read them when parsing.

What do you think?

olivier-stasse, 2016-03-02:
Can you develop the rational behind your assumption that $q_min$ is the function of ${\bf q}$ ?

aelkhour, 2016-03-02:
@olivier-stasse I don't know any example for position limits, but joints which are coupled through a transmission (e.g. a differential train) can have velocity and torque limits which are dependent of the position.

@jcarpent Could you explain more why this option is not satisfying?

jcarpent, 2016-03-02:
As mentioned by @aelkhour , the velocity and torque limits can depend on the current configuration. But it is not the case for the position limits.

I find this solution a little bit painful. 
But I would suggest to make velocity and torque limits dependant of the configuration, and to keep the other constant regarding the configuration.

aelkhour, 2016-03-02:
But how can you be sure that position limits are constants for all types of mechanisms?

nmansard, 2016-03-02:
I am very against the current implementation, that I think complex for 
nothing. Indeed, we have yet no strategy to have limits depending on the 
configuration, and no clear idea about how it may be useful. Taking for 
example the case study reported by Antonio: differential drive: this is 
not a matter of joints, but a matter of actuation (and we have yet not 
actuator models in Pinocchio).

I deeply believe that it is a bad procedure to imagine an API before 
understanding how the API should be used. Here we have no clear idea of 
how it should be implemented: then implement it simply. We will do 
something more complex when we will really need it (at very likely, it 
will be done differently to the current implementation).

More practically. The limits should be implemented in the model: the 
joint limits are element of the configuration space (i.e. represented by 
vector of size nq). The velocity and torque limits are element of the 
joint velocity space (~and its dual, ie size nv). When the joint is 
unbounded, the corresponding value is inf. No other implementation is 
adding value with respect to this simple solution.

Now, more complex limits might be desirable, function of the 
configuration or coupling components of the configuration. But then, 
they are no "bounds", i.e. of the form x <= xsup, but rather of the form 
Ax <= asup, or something similar. If this solution is necessary, we 
should think to a way of storing them (but not in the joint limits).

N

olivier-stasse, 2016-03-02:
I totally agree with the comment of Nicolas.

aelkhour, 2016-03-03:
Agreed @nmansard

fvalenza, 2016-04-01:
Solved with #141

closed 2016-04-01 by fvalenza