stack-of-tasks/pinocchio #119: Check jointDense for all the joints

jcarpent, 2016-03-03:


fvalenza, 2016-06-15:
Do we still need jointDense with the new JointAccessor ?

nmansard, 2016-06-15:
Yes indeed. This has a different purpose (enabling virtual inheritance 
for object-oriented developers).

jcarpent, 2016-06-16:
An other question is: do we need jointDense converter for all joint types ? 
I do not think so as it is just a copy of the sparse quantity to their dense counterpart, while joint accessor does the same job but avoiding copy.

nmansard, 2016-06-16:
I agree with that: we do not need conversion any more.
However, would you also agree that now, automatic conversion should be 
possible (from JointVariant to JointDense)?

jcarpent, 2016-06-16:
With the new JointAccessor, it is now possible to get the conversion of the sparse spatial quantities to their dense representation.

jcarpent, 2016-06-28:
No need this feature in the very near future. Consider this Issue as solved.

closed 2016-06-28 by jcarpent