stack-of-tasks/pinocchio #126: Clean Model

jcarpent, 2016-03-05:
Currently a model is build from concatenation of bodies that are linked one to each by joints. We need to rethink this strategy by appending bodies to a joint tree.

This new factorisation should help the merge of two models for example, it should simplify the URDF parsers, etc.

jmirabel, 2016-03-05:
It could even be a HPP requirement as it is needed for the manipulation stack.

My experience of URDF/SRDF parsing in HPP leads me to thinking :
- the URDF parser should create a joint tree below a root joint provided as input,
- parsing several SRDFs for the same robot makes writing SRDF easier (no need to copy SRDF to add new data)

jcarpent, 2016-03-06:
Thank you for your feedback. Can you say more about what you need for hpp-manipulation ?

nmansard, 2016-05-12:
The issue raises two aspects: there is a lot of members std::vector<something> in Joint; and the API is not nice, especially the method addBody (which indeed added both a joint and a body).

It seems that the proper solution is to keep the same members in Model, but modify / rename the API.

@fvalenza will separate the actual "addBody" method into "addJoint" (with zero inertia) and "appendBodyToJoint".

Should we keep a method "addJointAndBody" which calls both? I think yes.

jcarpent, 2016-06-15:
Solved by #184.

closed 2016-06-15 by jcarpent