stack-of-tasks/pinocchio #145: eigen compiltation issue

lengagne, 2016-04-07:
Hi,

I am trying to compile pinocchio, with eigen 3.2.0 (install from ubuntu packages) and I got 
the following error : 

/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:243:33: **error: no matching function for call to** ‘Eigen::Matrix<double, 3, 3>::_init1(const se3::Symmetric3Tpl<double, 0>&)’
       Base::template _init1<T>(x);
                                 
and I got several candidates as presented in the attached file

[make_log.txt](https://github.com/stack-of-tasks/pinocchio/files/208388/make_log.txt)

Maybe I need to install earlier version of Eigen ?

Can you help me ?

jcarpent, 2016-04-07:
Hi Sébastien,

Can you precise your version of Ubuntu?
I think your version of Eigen is not 3.2.0. I checked on eigen src, and the line 243 of Eigen/src/Core/Matrix.h looks like:
>      Matrix(Index rows, Index cols);

Can you check it in /usr/local/include/eigen3/Eigen/src/Core/util/Macros.h ?

For sure, pinocchio works perfectly on 12.04, 14.04 and 15.10, with Eigen starting from 3.0.5. If you want to install it automatically, you can follow the install description: https://github.com/stack-of-tasks/pinocchio/wiki/Installation

lengagne, 2016-04-07:
My version is Ubuntu 14.04.4 LTS.

I checked, indeed I have two version of Eigen and the compiler chose the one I installed manually. It is version 3.2.90 (I attached the Matrix.h and Macros.h files of Eigen)

[Eigen.zip](https://github.com/stack-of-tasks/pinocchio/files/208436/Eigen.zip)

jcarpent, 2016-04-07:
Ok. In fact 3.2.90 is not consistent with the current API, that's why your version does not compile.
You should use 3.2.8, the last released version.

lengagne, 2016-04-07:
Hi, I removed eigen3.2.90 and it compiles.

Now, I just have to use it ;)

closed 2016-04-07 by lengagne

jcarpent, 2016-04-07:
Ok, enjoy !