stack-of-tasks/pinocchio #146: acces to optimized code

lengagne, 2016-04-07:
Hi,

I succeed to make the benchmark work.

Is it possible to have access to the optimized code that is generated by pinocchio
(not the generated library, but the c/c++ code that is generated) ?

I cannot find any *.c or *.cpp  files that are generated.

Thanks

nmansard, 2016-04-07:
There are none. The efficiency of the library comes from the 
template-based programming, mostly implemented in the joint classes.

Maybe you mismatch with Metapod (the beta version of Pinocchio). I think 
you will find Pinocchio more comfortable to use, while the performances 
are quite similar to Metapod.

jcarpent, 2016-04-07:
To be more consistent, Pinocchio exploits metaprograming and implements similar tricks than Eigen (but with maybe less grace). For further details, there is a paper (https://hal.archives-ouvertes.fr/hal-01122475) explaining the way Metapod works (and in some points, Pinocchio).

And, if you install EigenPy (a binding between Eigen and Numpy - https://github.com/stack-of-tasks/eigenpy), you can also use Pinocchio with Python. You have also tutorials to learn the API:

1. Build a robot: http://homepages.laas.fr/nmansard/teach/robotics2015/1-build-your-robot.html
2. Graps an object: http://homepages.laas.fr/nmansard/teach/robotics2015/2-grasp-an-object.html
3. Static walk: http://homepages.laas.fr/nmansard/teach/robotics2015/3-static-walk.html
4. Dynamic walk: http://homepages.laas.fr/nmansard/teach/robotics2015/4-dynamic-walk.html

Finally, in the build directory, you can do `make doc` to produce the current documentation (which is only partial but will be updated in the near future).

olivier-stasse, 2016-04-08:
On rather new CPU pinocchio is faster than Metapod (this is why I decided to stop development on Metapod), The gain on CPU such as ATOMs than you can find in Romeo is marginal compare to the overall cost, and the rigidity of Metapod,
The biggest advantage of Pinocchio over Metapod is the fact that you can change your model on line.
Metaprogramming has been kept only in the relevant part by @nmansard . Code generated automatically may impact the performances by preloading data and code from the memory (as specified by Featherstone in his book). I believe Pinocchio is currently the best solution that we have tried so far.

lengagne, 2016-04-08:
Ok, so

If I want to have optimized code (not optimized execution time), do I have to use at Metapod ?

Does pinocchio is able to deal wth template for variables q,qdot, ..., in order to consider other type than double ?

Thank you for your help.

jcarpent, 2016-04-08:
Currently, double is the default Scalar type. A future PR will be proposed by @fvalenza to define a scalar_t which can be either float, double, etc. This will be very useful for Romeo or Nao because of Atom processors.

What do you mean by optimised code? What is your purpose?

olivier-stasse, 2016-04-08:
The point is that pinocchio has been optimized for nowadays CPU (typically Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz with 6Mo of L2 cache) .
As I said before loops may be much more efficient that loop unrolling due to the time taken to fetch data and code from the memory.
It is a codesign approach: the code fits the hardware it is running on.

lengagne, 2016-04-08:
"the code fits the hardware it is running on." 
indeed my first goal is to do the same but with different hardware (shh it is secret).
So we would like to have access to the matrix and vector of the inverse dynamic model.

My second goal is to use pinocchio with automatic differentiation and interval classes.

nmansard, 2016-04-08:
Metapod is generating code to describe the robot, not dedicated code for 
the algorithms. It will not be more useful to you than Pinocchio.
For automatic generation of algo loops, you have to dive inside the old 
versions of JRL-dynamics and Humans by PBW.

Pinocchio has two parts. The first one implements the spatial algebra of 
Featherstone. The second part implements the robot model and the 
algorithms to compute model values, using the first part.

The first part is templated, and should be able to handle automatic 
differencing, although it has been hardly tested exept for double (and 
maybe float). However, you have some strong unittests that you can 
extend to automatic differenciation.

The second part is not templated but only works for double. It would be 
useful to template it. However, we do not have the manpower yet to do 
so. We would be happy to support your team if you volunteer to do so. 
Moving from double to float should not be a big effort. Possibly, moving 
to more exotic types, like complex for automatic differencing, may be 
more tedious.

With respect to your final objective of finite differencing, I would 
suggest to start from the spacial algebra implemented in Pinocchio, and 
implement a dedicated version of RNEA. This is not a long term solution, 
but is doubtlessly the simplest to implement. I can guide you to 
re-implement RNEA is need be.

lengagne, 2016-04-08:
Hi,

I am interested to add template to pinocchio. I am not sure I have the abilities but it is something interesting in the future.

For the moment I will have a look at humans and compare pinocchio with my template implementation of robotics algorithm (modified from the work of Martin Felis)

nmansard, 2016-05-12:
@jcarpent is about to push a cleaner version of spatial algebra, where it should be possible to do automatic derivation. We are not planning to template the joint yet.

closed 2016-05-12 by nmansard