stack-of-tasks/pinocchio #161: Update python viewer

jcarpent, 2016-05-04:
Since we are now able to save the current visual info, we need to move forward by loading directly the visual geometries inside the gepetto viewer. And then, the URDF module inside the gepetto viewer will become deprecated.

jmirabel, 2016-05-04:
IMHO, tagging the URDF module of the gepetto-viewer as deprecated is not a good idea as I don't think it is a good idea we can remove it in a near future. It would break the ability of loading data in a remote viewer, with paths to the files being different.*
Keeping the path in ROS format (`package://`) should work though.

PS: if you wonder who is using a remote user: I am using it when I run benchmarks on a faster remote computer

jcarpent, 2016-05-09:
For sure, what we can do is to use the resource retriever of ROS or Pinocchio for the remote files. But in a near future, I would suggest to remove the strong dependency of URDF parser from Gepetto Viewer and to work only with our own description of kinematics chain for generic and general purposes.

nmansard, 2016-05-12:
Pinocchio will stop using the URDF interface of Gepetto-Viewer and call directly Gepetto-Viewer with addMesh(), to build single solid objects.

I think that moving URDF_DOM as optional dependency of Gepetto viewer would be nice, as a very optional feature.

jmirabel, 2016-05-12:
I'm not against removing URDF_DOM dependency but:
- I think it is not worth to spend time making it optional if the goal is to remove it later.
- Pinocchio is not the only software using Gepetto-Viewer, so we cannot design the viewer to fit Pinochio requirements at the cost of losing compatibility with other softwares. I do think this discussion belongs to that repository.
- I don't think this can happen before a merge between Pinocchio and hpp-model (at least, because I do not know who else is using it).

The idea of removing the pain of (poorly handled, at the moment) handling offsets between links and meshes of URDF is very pleasant to me ! It would solve some other dependent bugs. It also means Pinocchio has to handle those offsets.

nmansard, 2016-05-12:
I fully agree with this last message. Yet, let simply make Pinocchio work without the URDF module of Gepetto-Viewer, without touching Gepetto-Viewer.

jcarpent, 2016-06-15:
Solved by #184.

closed 2016-06-15 by jcarpent