stack-of-tasks/pinocchio #171: Compiling problem with geomId

olivier-stasse, 2016-05-20:
Hi,
Sorry I did not found geomId in geometry.hpp called here:
https://github.com/stack-of-tasks/pinocchio/blob/master/unittest/geom.cpp#L430
Is this a missing commit ?

jcarpent, 2016-05-20:
No. This unit test is compiled with you have HPP and either your version of HPP is too old or this unit test must be updated to fulfil the last version of HPP.

fvalenza, 2016-05-20:
The method name "geomId " may have changed in a previous rework of the way
a geometry model store the geometry objects. I do not have my laptop to
check it now but to solve it now it could be  "getCollisionId" or
"getVisualId"

this needs to be updated in the optional tests indeed

jcarpent, 2016-05-24:
Solved by #173.

closed 2016-05-24 by jcarpent