stack-of-tasks/pinocchio #175: URDFDOM release 0.4.1

olivier-stasse, 2016-05-27:
Dear all,
A recent test in compiling pinocchio with urdfdom release 0.4.1 was a failure.
The population of the data structure is different and I had to downgrade urdfom to 0.3.0 to make pinocchio works.
You can expect problems with newer ROS releases.

aelkhour, 2016-05-30:
That's strange I did not have this issue on Ubuntu 16.04 and urdfdom 0.4.1, all pinocchio tests are passing. I'm using pinocchio release 1.1.0, didn't test dev branch.

jcarpent, 2016-05-30:
Did you have any problem when installing Pinocchio on 16.04? I suppose you did install a release version of Eigen (3.2.8) and you are not using the one coming with 16.04?

aelkhour, 2016-05-30:
@jcarpent Indeed I followed the install recommendations and installed Eigen 3.2.8. I didn't even try to build pinocchio against the system version.

olivier-stasse, 2016-05-30:
Thas was indeed on devel branch.

jcarpent, 2016-06-22:
I did not have this issue on OS X with urdfdom 0.4.1 neither with the latest release 1.1.2 of pinocchio or the devel branch.

jcarpent, 2016-06-27:
@olivier-stasse Do you sill have the problem?

jcarpent, 2016-07-06:
I have tried on Ubuntu LTS 12.04 with console-bridge 0.3.0 and urdfom* 0.4.1 and everything goes well. The only thing is console-bridge requires c++11 for compiling, but no more.

closed 2016-07-07 by jcarpent