stack-of-tasks/pinocchio #179: Fixed Joints Support

KevinAStein, 2016-06-08:
Hi,
I am trying to load a urdf model to Pinocchio. Looking at robotModel->bodyNames() I can see that there not all bodies specified in the URDF file are listed.
I have specified some fixed joints in the chain from the root link to the feet. Those seem to be missing. Is it not possible to have fixed joints inside a kinematic chain?
thanks,
Kevin

fvalenza, 2016-06-08:
Hello Kevin,
In Pinocchio, fixed joints are handled this way :
We don't add a Joint of type "Fixed" in the Model but we add a Frame at the center of the fixed joint ( so that we keep track of the information ) and we merge the inertial information of the child link with the parent of the fixed joint.
If you have Joint1 -> link1 -> fixed Joint -> link2, it will be parsed as Joint1 -> link1 + link2 ( expressing the inertial informations in the good frame).
It is then normal that you can't see it in bodyNames(), you can however find it in robotModel-> fix_bodyNames

KevinAStein, 2016-06-08:
Thanks for your answer.
I have something like joint1 -> link1 -> fixed joint -> link2 -> fixed joint -> link3 and want to compute kinematics and dynamics for link3. How can I access this link? existBodyName("link3") and getBodyId("link3") lead to an error.

fvalenza, 2016-06-14:
Sorry for the delay.
It depends on which version of Pinocchio you are running:
Release/Master Branch
----
It can be tricky as the link3 as an object disappear during the parsing. Hence you have to work with the "fixed joint". But Pinocchio doesn't have fixed joints. As stated, what it does is it creates a Frame whose name will be the name of the fixed joint, attach it to the parent joint of the fixed frame with a relative placement. In your case you will have a frame instead of the _fixed joint 2_ ( = fixed joint supporting link2)  and a frame instead of the _fixed joint3_.
- Bodies attached to regular joints just "disappear" because we consider the entry point ( = the frame attached to the body) of the body to be the same as the center of the supporting joint

- So to acces the link you will have to do : getBodyId( getFrameParent( "FixedJoint3" ) )

Devel Branch ( coming in next pull request)
----
Bodies/Joints/Frames are currently being reworked.

- Each body will have a Frame (such as said for bodies attached to fixed joints). There will be redundant informations ( for bodies attached to regular joints, you may access it from an operational_frame or from the joint supporting the body)

- To acces the link, you will have to do: getFrame ("link3") and ensure the Frame is of type "BODY"

You can find in algorithms/operational-frames.hpp the algorithms currently available working with Frames. You can compute the forwardKinematics

KevinAStein, 2016-06-15:
Thanks for your help. The problem is that, in my case, the two fixed bodies at the end of my kinematic chain have a rotational and spacial offset to the last actuated joint. So using the FrameParent seems not to work.

fvalenza, 2016-06-15:
So the paragraph __Devel Branch__ of my previous answer is what you need (We can add an offset (i.e a framePlacement ) with respect to the parent joint for any Frame ) .
We may do the change soon

As a temporary solution, If you have only a few bodies like that, you can add Frames by hand using Model::addFrame

```C++
 ///
/// \brief Create and add a frame to the kinematic tree.  
///
/// \param[in] name Name of the frame.
/// \param[in] parent Index of the supporting joint.
/// \param[in] placement Placement of the frame regarding to the joint frame.
///
/// \return Return true if the frame has been successfully added.
///
bool addFrame(const std::string & name, const JointIndex parent, const SE3 & placement);
```

closed 2016-06-29 by nmansard