stack-of-tasks/pinocchio #188: The Geometry Module has not been compiled with Pinocchio

stephane-caron, 2016-06-17:
Hi there,

Trying to load the Romeo model:

    import pinocchio
    from pinocchio.romeo_wrapper import RomeoWrapper
    robot = RomeoWrapper(path_to_romeo_urdf)

I am getting the following exception (from robot_wrapper.py:35, called from romeo_wrapper.py:26):

    Exception: the Geometry Module has not been compiled with Pinocchio. No geometry model

However, I see no option to compile or not a geometry module in ``ccmake``. Did I miss something in the install process?

PS: I have ``urdfdom`` installed and found alright:

    -- checking for module 'urdfdom >= 0.2.10'
    --   found urdfdom , version 0.2.10

nmansard, 2016-06-17:
The error is telling you that you have no geometry support: basically, HPP-FLC (a conditional dependency of Pinocchio) is not installed. Pinocchio should work without. However, robot-wrapper is not tolerant enough.

This is bad. The integration of the Geom in the robot wrapper needs some clean up. I will try to do that today.

In the meantime, you may want to install HPP-FCL. I think you may have some interest in having geometry support in the future.
Alternatively, you can quickly edit robot-wrapper and remove or correct this bad test and the following assert.

jcarpent, 2016-06-17:
The solution is to suppress the exception raising in the init inside RobotWrapper.py file.
This is a bug that I introduced some time ago.

I will fix it now.

jcarpent, 2016-06-17:
solved by #189.

jcarpent, 2016-06-17:
Thanks @stephane-caron for your report.

closed 2016-06-17 by jcarpent

fvalenza, 2016-06-17:
Please keep in mind that all robotWrapper's built-in functions related to displaying in the viewer are dependant on the geometry module since it accesses the visuals

jcarpent, 2016-06-17:
@fvalenza So the best thing to do is to allow the reading of the visual and mesh files path but not their opening.