stack-of-tasks/pinocchio #195: Changing the API of the spatial module

nmansard, 2016-06-22:
This issue is open to discuss the branch "future", and more specifically the 7 commits composing the sub-branch topic/spatial. 

The purpose of this branch is to propose some modifications of the spatial module (mostly composed of the classes SE3, Motion, Force and Inertia). The objectives of the change are listed below:
* Introduce more consistence in similar classes SE3, Motion, Force and Inertia.
* Improve the performance of the RNEA and other main algorithms.
* Strengthen the rigour of the code with respect to the mathematical structure of the objects.

The changes are yet only partial. Some modifications have been prototype but not yet systematically applied. The branch is functional. However, additional work is needed to make the changes consistent. Since additional work is needed, I propose to discuss here the list of changes and establish which changes are necessary and should completed, and which changes could be kept for later or even discarded.

I propose to discuss each commit after the other (which is why I refactorize the branch). First, let's discuss commit 19e1f1 ([C++] Spatial: modify API) in the next message.

nmansard, 2016-06-22:
I discuss here the two first commits of the branch topic/spatial: 5bcbce ([C++] Spatial: add SpatialBase) and 19e1f1 ([C++] Spatial: modify API).

In a first time, I propose to make an exhaustive list of the changes in the API module. We can then discuss the relevance of each changes.

1. Introduce an additional API level with a CRTP class SpatialBase. This CRTP API imposes a common behaviour in all the classes of the module, in particular the presence of the SE3Action and SE3ActionInv methods.

2. Change the name of operators. C++ operators are only trivial inline methods, while the code is in methods with an ad-hoc name. The name was yet taken from Python standards (for example __plus__ for operator+, __eq__ for operator==). They are now renamed following Eigen standards (add for operator+, isEqual for operator==).

3. All the methods (mostly operators) of the spatial modules are now template. Take the example of the operator+ (__plus__) of the motion class, taking in input two motion objects and returning a motion object: A <- B+C. This operator was only working with two motion of the same type and return again the same class (i.e. A, B and C are of the same type). Now C is templated (A should still have the same type as B).

4. Among the operators, emphasize the importance (efficiency) of operators += and -= versus operators + and -. In particular, A = B+C+D would be less efficient than A=B; A+=C; A+=D. See point 6 below.

5. The four main classes are implemented in three main parts, composed of (i) the base class imposing the API, (ii) the dense class, post-fixed by Tpl (SE3Tpl, MotionTpl, ForceTpl, etc) implementing the base class, and (iii) specific implementation corresponding to the joint, often exploiting the structure (sparsity) of the object to improve the efficiency of the computation.
It is proposed to add an additional CRTP layer to implement default behaviours for the classes of the group (iii), so that safe behaviour (either clean compilation error, or default (suboptimal) run-time operator) even if all the methods of the API are not implemented in the specialized classes.

6. Introduce a computation tree, i.e. when an operation A = B+C+D is interpreted, first build a trivial object representing the tree of computation (in the example something like OpPlus< B, OpPlus< C,D > >), which is only explicitly computed when interpreting the operator=. This computation tree follows Eigen standards.

7. InertiaBase is too much biased toward the typical mass-com-inertia implementation (InertiaTpl), which is indeed sparse (then dense version of Inertia would be a 6x6 symmetric matrix). InertiaBase has been consequently simplified to avoid this bias.

8. Change the names SE3Action and SE3ActionInv (sometime suffixed by _impl) by SE3ActOn and SE3InvActOn. 

9. Suffix all CRTP method implementation by _impl, to avoid infinite loop (stack overflow) at run time due to the absence of proper implementation.

I propose now to open one issue for each of these 9 points, all started with the ttitle [Spatial.NN] with NN between 1 and 9. Possibly, I missed some important changes in these two commits. In that case, please add some points 10 ... in answer to this issue.

jcarpent, 2019-02-28:
This issue has been partially solved by Pinocchio 2.0.0

closed 2019-02-28 by jcarpent