stack-of-tasks/pinocchio #196: [Spatial.9] Suffix all CRTP method implementation by _impl ...

jcarpent, 2016-06-22:
I think it is a good strategy to avoid infinite loop and decrease debugging time.
I would also suggest to set all those methods protected to force the user to use standard C++ operators and to let the compiler solve the computation tree.

nmansard, 2016-06-23:
I agree that this is necessary. We should have a single commit to propagate this change systematically in all the code (mostly spatial and joint modules). Anybody would volunteer to do it?

fvalenza, 2016-06-24:
If nobody has started it, i can do it

jcarpent, 2016-07-06:
I just realised that it is not possible to set methods with *_impl suffix protected. Or, one should use some back door to do that which leads to unreadable code.

In Eigen, they used some tricks to compile an API which fits pretty well for doxygen but which does not correspond to the real API, in terms of implementation. But I think the code will become very hard to read and maintain to.

Maybe the best thing for a clean API is to resort on gdb during the debug process ...