stack-of-tasks/pinocchio #197: [Spatial.7] InertiaBase is too much biased toward the typical mass-com-inertia implementation (InertiaTpl) ...

jcarpent, 2016-06-22:
`InertiaDense' is mostly used inside the ABA algorithm. It can be represented with different memory representation:

- Symmetric 6x6 matrix. Which takes 36 doubles in memory and are not really efficient in Eigen.
- 2 Symmetric 3x3 matrices for the diagonal part and one Dense 3x3 matrix for the remaining block. This solution has to be tested.

Other strategies can also be thought.

nmansard, 2016-06-22:
Could you quickly recall the shape of nondense inertia? Is it?
[ m I_3    R
   R^T        A  ]
With I_3 an identity, A symmetric and R any 3x3 matrix. In particular, R is not in general anti symmetric, is it?

jcarpent, 2016-06-22:
The shape is much weird:
[A     B
 B^T C]
with A and C symmetric 3x3 and B any 3x3 matrix.

nmansard, 2016-06-23:
So, let s do it. 

Shall we discuss the naming? The standard spatial inertia is (m,c,I). I think this implementation should have the same naming standards as MotionTpl and ForceTpl (although it is a little bit sparser than MotionTpl and ForceTpl). 

What about InertiaTpl and InertiaABA?

nmansard, 2016-06-23:
BTW, should we rename MotionTpl, ForceTpl and SE3Tpl as MotionDense, ForceDense and SE3Dense?

jcarpent, 2016-06-23:
I would go for InertiaSparse and InertiaDense (because InertiaABA is also templated).
I'm also in favour of renaming ClassTpl into ClassDense and using the Sparse suffix for their counterparts.