stack-of-tasks/pinocchio #198: [spatial.1] Intriduction of SpatialBase

nmansard, 2016-06-22:
Introduce an additional API level with a CRTP class SpatialBase. This CRTP API imposes a common behaviour in all the classes of the module, in particular the presence of the SE3Action and SE3ActionInv methods.

nmansard, 2016-06-22:
Background: SE3 objects can "act" on all objects of the spatial modules (on motion, forces, inertia and also on SE3 itself). In the last release, acting is done using a template method of SE3 named "act" which in turn calls the method SE3Action of the class on which the action is applied. Eg se3.act(motion) calls the method motion.SE3act(se3).
If applying se3.act on an object A that does not have the SE3action method, then an error an compilation is raised stating that A does not have a SE3action method.

The error would be more clear by forcing the method act to only apply on classes implementing a CRTP interface documenting the presence of a SE3action method. I.e. 
Template<D> ReturnType SE3::act( SpatialBase<D> & obj) { return D.SE3action(*this); }

Pro:
- the compilation error is easier to read.
cons: 
- the current compilation error is not too difficult neither 
- an extra crtp level is added, making the module more difficult to understand.

nmansard, 2016-06-29:
In the current state, SpatialBase would clarify the API and make it more rigorous but would not bring any added functionality. 

SpatialBase would be very useful if trying to go beyond the current algorithms developped in Pinocchio. For example SpatialBase would help to make explicit the connection between the __act__ operator and the __cross__ operator (i.e. the fact that the derivative of m.act(Z) is m.act(dZ) + dm x m.act(Z), with m in SE3, Z a spatial object, dZ the derivative of Z and dm the spatial velocity corresponding to the derivative of m.

However, it would require a lot of work to make this link explicit. Until it is done, it is doubtful that SpatialBase would be helpful to the current API.

jcarpent, 2018-10-31:
This was in fact not necessary for Pinocchio 2.0.0

closed 2018-10-31 by jcarpent