stack-of-tasks/pinocchio #200: [spatial.3] adding template on operators

nmansard, 2016-06-22:
All the methods (mostly operators) of the spatial modules are now template. Take the example of the operator+ (plus) of the motion class, taking in input two motion objects and returning a motion object: A <- B+C. This operator was only working with two motion of the same type and return again the same class (i.e. A, B and C are of the same type). Now C is templated (A should still have the same type as B).

nmansard, 2016-06-22:
Take again the example of operator+ of the motion class, for any implementation say MoD (MoD is for example a MotioonRevolute). Currently the API only imposes MoD to sum with other MoD objects.
MoD MoD::operator+(MoD& m2) 
If trying to sum MoD with another motion type, for example with MotionTpl, it would not go through the API but by defining an external operator+ with
MotionTpl operator+( MoD&, MotionTpl&)

What is not satisfactory with this? Essentially, it is working but without proper excited specification, for example through the API. But more difficult, it would not work for methods not associated with operators. For example, the dot product is called with m.dot(f), and cannot be implemented with an external operator outside the CRTP API.

nmansard, 2016-06-29:
We will add the two following operators in the Base API:
MotionDense MotionBase<D>::__add__ (MotionDense)
MotionBase<D> MotionBase<D>::__add__ (MotionBase<D>)

A priori, it is not necessary to further add operations with any other implementation. If it is necessary, the third line to be added in the API would be:
template<D2>
traits<D>::AddReturnType<D2> MotionBase<D>::__add__(MotionBase<D2)
It is unlikely that adding this to the API is useful. Let's not add it yet.

jcarpent, 2018-10-31:
This feature is now covered by Pinocchio 2.0.0

closed 2018-10-31 by jcarpent