stack-of-tasks/pinocchio #201: [spatial.4] In-place operators

nmansard, 2016-06-23:
Among the operators, emphasize the importance (efficiency) of operators += and -= versus operators + and -. In particular, A = B+C+D would be less efficient than A=B; A+=C; A+=D. See point [Spatial.6].

nmansard, 2016-06-23:
Nothing specific to do here. I added this issue to remember that these operators should be propagated in the algorithms.

An alternative would be to implement a complete computation graph (see Spatial.6), which would then make operator+ as efficient as in-place operators.

I think the easy option (use += instead of +) should be implemented first.

jcarpent, 2016-06-23:
I think the main question here is about how to encode the fact that for Motion and Force, the group operator is commutative. We then have to use the best strategy for the `operator+` and `operator-` in the context where one of the arg is Sparse and the other one is Dense.
In this case, we may have a default behaviour in the Sparse class which will call the related operator of the Dense class.

jcarpent, 2018-10-31:
Some of the computations are now in-place operators since Pinocchio 2.0.0

closed 2018-10-31 by jcarpent