stack-of-tasks/pinocchio #202: [Spatial.5] Introducing an additional CRTP level for sparse implementations

nmansard, 2016-06-23:
The four main classes are implemented in three main parts, composed of (i) the base class imposing the API, (ii) the dense class, post-fixed by Tpl (SE3Tpl, MotionTpl, ForceTpl, etc) implementing the base class, and (iii) specific implementation corresponding to the joint, often exploiting the structure (sparsity) of the object to improve the efficiency of the computation.
It is proposed to add an additional CRTP layer to implement default behaviours for the classes of the group (iii), so that safe behaviour (either clean compilation error, or default (suboptimal) run-time operator) even if all the methods of the API are not implemented in the specialized classes.

nmansard, 2016-06-23:
In the CRTP design, there is no equivalent notion to "pure" virtual in classical inheritance. Consequently, we cannot force the class inheriting from CRTP base classes to implement all the methods of the API (e.g. in MotionBase, a method "linear" returns the linear part of the velocity; however, none of the specific implementation like MotionRevolute are implementing this method).

When a method of the base that is not implemented is called by a third party, there is a compilation error like below
motion.hpp:45:75: error: ‘se3::MotionBase<se3::MotionRevolute<0> >::Derived_t’ has no member named ‘linear_impl’
     Linear_t linear()   { return static_cast<Derived_t*>(this)->linear_impl(); }

Mainly, the CRTP base is declaring all the methods that will be implemented by the dense implementation (i.e. MotionTpl for MotionBase). However, the other classes implementing the CRTP base (i.e. MotionRevolute --and the other motion for joints-- for MotionBase) are typically implementing only the methods that are necessary in the context they are defined for.

As we know that most of the classes implementing the CRTP base will not implement all the methods of the API, we may want to anticipate the lacks, by providing an additional CRTP level from which these partial implementation would inherite. This additional CRTP level would provide a default behavior. A prototype of such an additional CRTP level is proposed for motion, named MotionSparse. The inheritance tree is:
* MotionBase is the basic level.
* MotionTpl (i.e. the dense implementation) inherits directly from MotionBase
* MotionSparse inherits from MotionBase. No link with MotionTpl
* All the other motions (dedicated for joints) inherit from MotionSparse.

The two questions are then:
Q1. what should be the default behavior. 
Q2. Is it worst spending time and complexity to implement this selected default behaviour.

nmansard, 2016-06-23:
For Q1, I am in favor of implementing an explicit compilation error as default behavior.

For Q2, I would say that adding code to have a better compilation message is not worth the effort, knowing that the initial error message is not bad. I am a little bit afraid of the extra complexity brought by the additional CRTP level.

jcarpent, 2018-10-31:
This is not needed either. Other approaches have been followed in Pinocchio 2.0.0.

closed 2018-10-31 by jcarpent