stack-of-tasks/pinocchio #203: [Spatial.6] Computation tree

nmansard, 2016-06-23:
Introduce a computation tree, i.e. when an operation A = B+C+D is interpreted, first build a trivial object representing the tree of computation (in the example something like OpPlus< B, OpPlus< C,D > >), which is only explicitly computed when interpreting the operator=. This computation tree follows Eigen standards.

nmansard, 2016-06-23:
A prototype of computation tree is implemented for SE3 in se3.hpp with SE3_product.

This feature is very interesting. There is a clear gain to expect. However, it is also a big piece of code to write. It is very important to carefully think about the design before starting to implement it. 

Additionally, it would be interesting to try first to connect with the computation graph of Eigen, possibly by inheriting the elements of the spatial module from MatrixBase.

If somebody is very interested by this topic and to lead a work task, I would be very motivated to follow the work group. However, I am afraid that we lack of development force yet to start it.

jcarpent, 2016-06-23:
For this point, I think we can at least implement it for the class Motion and Force to overload A = B + C + D  and using the Eigen already implemented strategy. 
As mentioned by @nmansard, we can directly inherit from Eigen::Matrix but it will only work for MotionDense. 

While for SE3, I think the gain can be good but we still need to allocate temporary result variables.

An other advantage of saving the computation tree is for automatic differentiation.