stack-of-tasks/pinocchio #208: Static allocator for model and data

jcarpent, 2016-06-24:
For critical systems, it may be useful to pre-allocate buffers to avoid dynamic allocation.

nmansard, 2016-06-24:
With more details: it would be good to have the sizes (nq, nv, nbodies) of model templated, either directly (e.g. nq being a template parameter of the Model class) or with a max bound templated. Consequently, the vectors representing q (Vector_Q, size NV) and vq, tauq (Vector_V, size NV) would be templated as well, either directly or with a max bound. This modification would be propagated in the algorithms. 

This modification would also to work with statically-allocated memory (garantee of no new operator at run time). Such a property is requested for critical systems, typically implied to work with industrial robots.

No roadmap to implement this feature yet. The work to be achieved is quite limited (1 week of work) and well defined. If anybody is interested either by the feature or by implementing it, just mention it.

jcarpent, 2016-07-22:
@aelkhour Can you propose some allocators you implemented for that issue?

jcarpent, 2019-02-28:
@aelkhour Do you still need this featue?

aelkhour, 2019-02-28:
No you can close it.

closed 2019-02-28 by jcarpent