stack-of-tasks/pinocchio #209: Remove the duplicata collision / visual from geometry

nmansard, 2016-06-27:
Syntactically, visual and collision objects are the same. It is only the semantic (brought by the end users) that makes visual and collision different. I therefore suggest to preserve this similarity as long as possible. This will avoid duplication of code. 

In geometry.hpp, remove notions of collision and visual. A ``GeometryModel'' object should then have a list of GeometryObject, without type (COLLISION / VISUAL) and a list of collision pair. In case the GeometryObject will be use to store visuals, this list of pair will simply be empty.

jcarpent, 2016-06-27:
We can add an enum with type VISUAL and COLLISION just to distinguish between both. It is just a matter of memory. In most of URDF files, COLLISION are lighter than VISUALs and one may prefer to use COLLISION objects for collision detection.
I propose to automatically load COLLISION objects and not VISUAL.

nmansard, 2016-06-27:
About loading: this is a matter to be handled in the parsers, typically 
in parsers/urdf.hpp. There, an argument should be added in buildGeom to 
specify if we are loading the collision or visual objects. Possibly, we 
may want that this argument is set to a default value "collision", also 
I would rather recommend to force the user to be explicit with this 
argument. No strong opinion there.

To be exhaustive: a complete model of the robot would be composed of 5 
objects:
* model of type Model
* data of type Data
* collisionModel of type GeometryModel
* collisionData of type GeometryData
* visuals of type GeometryModel
The last object "visualData" is useless and would typically not be 
created, also it is armless.

jcarpent, 2016-06-27:
I totally agree with the way to compose a complete model. This means also that we have to mostly rewrite the URDF geometry parsers. 

The main problem that I see is about the delayed process we have chosen: we load geometries after building the model where some reductions (due to fixed joints) may occur. We must then rely on model.frames to retrieve the near joint parent in the tree.

jcarpent, 2016-06-27:
We must also fulfil the requirements of #190.

nmansard, 2016-07-19:
Solved by #227

closed 2016-07-19 by nmansard