stack-of-tasks/pinocchio #211: frame.hpp should not be in spatial/

nmansard, 2016-06-27:
The header would be best in multibody/. This question is also related to the many minor .hpp and .hxx files that are cluttering the source directories. Should we have .../details/ directories to hide details (e.g. multibody/details/)?

jcarpent, 2016-06-27:
I'm in favour of adding the details folders just a matter of organisation.

fvalenza, 2016-06-27:
Concerning the .hpp and .hxx files.
Currently their use is more like one big file that is splitted in two and in file.hpp we do an `#include <file.hxx>` but I had a problem with JointComposite about circular dependencies where i have to remove this include at the end of my jointcomposite.hpp and add an `#include <jointcomposite.hpp` at the beginning of the jointcomposite.hxx file. It was like a normal (include)header/source file. So it was more than just "implementation details"

jcarpent, 2016-07-07:
Solved by #222.

closed 2016-07-07 by jcarpent