stack-of-tasks/pinocchio #212: Reorganize urdf parsers

nmansard, 2016-06-27:
First, we start to have quite some code in multibody/parsers. It might be a good idea to move this directory one level above, in src/parsers. Let's assume this solution for the following.

The API of URDF parser should only expose two methods: buildModel and buildGeom. The rest of the API are details, and should not be displayed to users and stored in a details namespace (se3::urdf::details).

I suggest that the code about URDF should be organized in 4 files:
parsers/urdf.hpp  /// the API, containing the two functions listed above.
parsers/urdf/model.hxx /// the code for parsing urdf tree building the model.
parsers/urdf/geometry.hxx /// The code for parsing the urdf tree building the geometry model.
parsers/urdf/utils.hpp   /// The utils functions (e.g. convertFromUrdf functions) shared between both hxx)

For now, we must also keep parsers/urdf/assimp.hpp (yet from-collada-to-fcl.hpp), until ### is solved.

All the code in parsers/urdf/ should be put in se3::urdf::details namespace to avoid exposing it in the doxygen.

I do not like to have a magic behavior in buildGeom, by non-explicitly relying on ROS_PATH. I would suggest to expose the ROS_PATH function in src/tools/file-explorer.hpp, while mentionning in the documentation of buildGeom that these two methods should be typically called together.

jcarpent, 2016-06-27:
I also think that we must moved to the parsers to the src level.

As suggested by some users, I think we must simplify at most the parsing process. So calling the retriever of the ROS_PATH inside the parser is a good option to simplify the parsing process. We have to inform the user through the documentation and maybe add an option to use or not this feature.

In the same spirit as moving the parsers directory, I'd suggest to make a shared library with all the parsers inside libpinocchio.so/.dylib

nmansard, 2016-06-27:
I would suggest to keep:

buildGeom( Model & model, string urdffilename, vector<string> paths )

with the documentation of paths being:
// @param[in] paths     A vector containing the different directories, 
typically obtained from calling se3::rosPaths()

Alternatively, we can have a second function:
buildGeomUsingROSPath( Model & model, string urdffilename )
or
buildGeom( Model & model, string urdffilename, bool usingROSPath )

nmansard, 2016-07-19:
Solved by #227

closed 2016-07-19 by nmansard