stack-of-tasks/pinocchio #215: Change se3 namespace into pinocchio

jcarpent, 2016-06-28:
The namespace se3 is ambiguous and need to be relevant for the project. I suggest to rename it in pinocchio.

nmansard, 2016-06-28:
Fair enough. 

Should we keep se3 as a sub namespace for the spatial module (::pinocchio::se3) ?

jcarpent, 2016-06-28:
To be more generic, we can also use ::pinocchio::spatial for the spatial module.

fvalenza, 2016-06-28:
+1 for ::pinocchio::spatial (no confusion with SE3 ( i remember some people of the team asking me if it was the same).

olivier-stasse, 2016-06-28:
+1 for ::pinocchio::spatial

jmirabel, 2016-06-28:
+1 for ::pinocchio::spatial

nmansard, 2016-07-26:
I recently worked a lot with a third-party lib and came to quite love the short se3:: namespace. I think we should keep it.

jcarpent, 2016-07-27:
I do not think so. The replacement is easy in the few related projects. 
The namespace se3 is not clear at all.

jcarpent, 2016-09-21:
ping! hpp-devel branch is now merged into the devel branch. It seems that Pinocchio will be quite soon merged into HPP. I suggest to change the global namespace se3 into pinocchio(or pin which is shorter). Any remarks?

jcarpent, 2018-10-03:
I'm going to address this issue in the very next weeks.
The idea is to replace `namespace se3`by `namespace pinocchio` and to add a shortcut to `pinocchio`namespace with `se3` nickname.

jcarpent, 2018-11-09:
Solved by #588.

closed 2018-11-09 by jcarpent