stack-of-tasks/pinocchio #218: Neutral position in model?

nmansard, 2016-07-05:
Should we add a "neutral" position in model class?

jmirabel, 2016-07-05:
It is a convenient method but isn't used a lot. If you do not want this in Pinocchio, I would say it can be implemented as a joint visitor in hpp-model.

jmirabel, 2016-07-05:
Just for the logs: the command 
```bash
find $DEVEL_DIR/src $DEVEL_DIR/include -type f \( -name "*.cc" -o -name "*.hh" \) -exec grep -l "neutralConfiguration" {} \;
```
gives:
```bash
hpp-model/src/device.cc
hpp-model/src/joint.cc
hpp-rbprm/src/sampling/sample.cc
hpp-rbprm/src/rbprm-limb.cc
hpp-rbprm/src/ik-solver.cc
hpp-model/include/hpp/model/device.hh
hpp-model/include/hpp/model/joint.hh
```

jcarpent, 2016-07-05:
I think this feature is not mandatory as it relies on an extra SRDF dependency. And also, it will not be used inside the algo but from outside. I would suggest just to add the SRDF parsing of this variable inside the toolbox, not more.

jmirabel, 2016-07-05:
The "neutral" configuration does not come from the SRDF. It is defined by the joints only:
- (1,0,0,0) for SO3,
- (1,0) for unbounded rotation,
- (0...) for all other joints.

Joint bounds are not taken into account.

jcarpent, 2016-07-05:
Ok, it was not clear for me and I was confusing.
@jmirabel How such feature is used inside HPP?

jmirabel, 2016-07-05:
It is used to get a configuration coherent with the joint types.

jcarpent, 2016-07-05:
You can then use for the same task random_config?

jmirabel, 2016-07-05:
Maybe, but not sure.
And anyway, we should not think in how we are using it now. It should be added if:
- it is a useful feature (mostly for testing purpose),
- it fits with the architecture and model (see questionable point below).

I don't like the idea of mixing random configuration and neutral configuration. The most questionable thing is that it does not use the bounds. For bounded joints, it would make more sense to have the center of the interval.

nmansard, 2016-07-05:
We could add a neutral position in the Model class. It is cheap. Such a 
configuration is used by many downstream users (e.g. reference 
configuration in IK for regularization purpose). It is quite similar in 
spirit to limit configurations, already included in Model.
We could then set up default strategies in an algorithm in pinocchio to 
set up this configuration, while the interested user would parse it from 
an SRDF file.

jcarpent, 2016-07-05:
The question is how you define the neutral config, knowing that you may have bounded joints as mentioned by @jmirabel but maybe with no defined bounds (aka inf)?

My main regards is to define something coherent and generic, not something very specific to HPP.

jcarpent, 2016-07-05:
From your two views, I cannot make the distinction between:
- neutral config is a class property 
- or it is a method which calls the neutral config of all the joints and returns a config vector (this can take into account the change in the bounds for example).

nmansard, 2016-07-05:
In HPP (note that we do not care so much about what HPP is doing): 
neutralConfig is a member of the class, which is initialized by the 
joint class (e.g. 0 for revolute, (1,0) for unbounded, (1,0,0,0) for 
quaternion spherical).

In Pinocchio: I suggest to add neutralConfig as a member of Model.

We can then discuss its initialization. I suggest to have a default 
initialization done by each joint class, following the spirit of HPP. 
And to add one (or several) algorithms to set up a neutral configuration 
(e.g. center of bounded joints). It would then still be possible for a 
user to reset neutralConfig, for example from reference values in SRDF.

jcarpent, 2016-07-05:
It looks more CLEAR now!

florent-lamiraux, 2016-07-05:
The neutral configuration is usually used to define initial and goal configurations for a motion planning problem by modifying joint values by joint name access. This is convenient particularly to  set the same problem with various models of the same robot (full PR2, PR2 without caster wheels,...). Defining configuration by a vector in  a script is not robust to those changes of robot model. See https://github.com/humanoid-path-planner/hpp_tutorial/blob/master/script/tutorial_1.py from line 11 for an example.

nmansard, 2016-07-05:
Ok. If now clear and relevant, I switch the issue in a TODO mode. Ok for all?

nmansard, 2016-07-08:
solved by #225

closed 2016-07-08 by nmansard