stack-of-tasks/pinocchio #226: Reference on joint id in CollisionObject

nmansard, 2016-07-11:
There is a redundancy of information in the geometry model. We maintain both:
* a list of "inner" objects for each joints of the model (stored in a map indexed by joint id, which should be indeed a list)
* the id of the parent joint inside the CollisionObject.
Possibly, the model could be inconsistent if geom.objects[geom.innerObjects[J_ID]].parent != J_ID.

Is this redundancy useful? If not, let's skip it.

fvalenza, 2016-07-11:
For the second point : when adding a collision object, we reference at the same time this object in the inner objects list so if the user does not modify, we _should_ not encounter this case

Concerning the redundancy, as for each "object" that we have in pinocchio, each entity knows the parent it is attached to, that's pretty clear (child to parent relation).
The inner list can be seen as the parent-to-child relation. I don't know too much of all the packages of hpp but for example in hpp-core/collision-validation.cc there are methods that add/remove collision pair for a particular joint. Hence it access all the collision objects attached to this specific joint

fvalenza, 2016-07-11:
Moreover each collisionObject __must__ know its parent as when we update its position we look for the absolute position of the parent joint first such as:

oMgeom = oMparent * parentMgeom

(It's the same for frames)

So this redundancy seems ok

nmansard, 2016-07-21:
We should remove inner/outer Objects from Model. Possibly keep it in data. Lets discuss that somewhere else.

closed 2016-07-21 by nmansard