stack-of-tasks/pinocchio #231: Alignmend issues

jmirabel, 2016-07-18:
Hello Pinocchio developpers,

when running the unittest, I get the traditionnal Eigen error:
```
rnea: /local/jmirabel/devel/hpp/install/include/eigen3/Eigen/src/Core/DenseStorage.h:86: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 6; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
```

The error is raised when creating an object of type `JointDataRevolute`.

I have a 32bit computer. I guess there is no alignment issues on 64bit arch. Do you have support for 32bit architecture ?

jcarpent, 2016-07-19:
Currently the simplest way to solve the problem is to add `-DEIGEN_DONT_ALIGN_STATICALLY` inside the compiler option line.

jcarpent, 2016-07-19:
To be more precise, it seems that the problem comes from boost::variant which does offer explicitly an aligned structure of the stack. Two solutions exist:

- one is to rewrite from scratch the mechanism of variant
- the other one is to inherit from boost::variant and to create our own aligned allocator

jmirabel, 2016-07-19:
Thanks for having dug that out !

jcarpent, 2016-08-09:
Solved by #298.

closed 2016-08-09 by jcarpent