stack-of-tasks/pinocchio #244: Model: avoid redundant addJoint methods

nmansard, 2016-07-21:
In Model, we have several methods addJoint and addJointAndBody. All of them can be achieved equivalently by calling then nominal addJoint(parent,type,placement,inertia,names), followed possibly some few additional methods to complete the call. I would like to suggest to remove this redundancy:
A. either by simply removing the redundant methods, while only keeping the basic ones
B. or by, at least, calling the basic methods in the complex ones, while removing the code duplicate. 

I vote for A.

nmansard, 2016-07-21:
In one of the methods, neutralConfiguration is assigned to be the middle point of the joint range. If this behavior is important, I suggest to write a dedicated algorithm that force the neutral configuration to be the range median, explicitly and a posteriori.

nmansard, 2016-07-21:
This becomes a HPP reqiuirement: we do not have the same behavior. I will implement the above-mentioned proposition.

jcarpent, 2016-07-26:
Partially solved by #251.

jcarpent, 2016-07-26:
Entirely solved by #251.

closed 2016-07-26 by jcarpent