stack-of-tasks/pinocchio #252: Add data copy algorithms

nmansard, 2016-07-25:
The Data structure is isolated from the Model struct so that we can create several instances of it in various locations of our application. On the other hand, we might expect that the same basic computations (like forward kinematics) would be necessary at these several locations. Shouldnt we then copy the needed data, with a scenario like below:
* compute FK at location 1
* use FK at location 1 (for example for RNEA)
* copy FK data from location 1 to location 2
* use FK at location 2 (for example for ABA)

I suggest to add a bunch of algorithms in a file src/algorithm/copy.hpp to copy the data from a const Data & source to a Data & dest, with flags to specify what must be copied.

Any opinion?

jcarpent, 2016-07-25:
It looks interesting to me.

nmansard, 2016-07-26:
Solved by a direct push (commit 94a96febe3d)

closed 2016-07-26 by nmansard