stack-of-tasks/pinocchio #254: CollisionPair

jmirabel, 2016-07-25:
The current implementation of CollisionPair inverts the index to have them in logical order. It is not very convenient.
A user who wants to read the distance results to get the closest points has no way to know that the indexes have been inverted and that the closest points have to be inverted.

I seems this is used only to ensure that there is no duplicates in the list. If this is the case, I see two options:
- option A : the user must remember whether the indexes have been inverted,
- option B: add an operator== to CollisionPair which handle inverted pairs.

I'm in favour of option B because I do not like the fact that the inversion of collision pairs is done in the back of the user.

jcarpent, 2016-07-25:
Just  to be clear, I sum up your proposition B:
You want to suppress the forced ordering and have any ordering. And add an operator== which takes into account the two situations. Isn't it?

I also suggests to add a method to sort all collision pairs by first index value for example. Just to be efficient in the when calling algorithms.

jmirabel, 2016-07-25:
I mean this:
`ColPairA == ColPairB` <=> `(ColPairA.first == ColPairB.first && ColPairA.second == ColPairB.second) || (ColPairA.first == ColPairB.second && ColPairA.second == ColPairB.first) || `

About the method to sort, I agree. But I think the method should not be called automatically. Or, at least, ahoud be such that the user can disable sorting the pairs.

jmirabel, 2016-07-25:
I can do a pull-request with option B.

jcarpent, 2016-07-25:
Let's do that !

jcarpent, 2016-07-26:
Solved by #255.
Maybe we still need to check the Python interface.

closed 2016-07-26 by jcarpent