stack-of-tasks/pinocchio #256: Unit test : jacobian against integrate

nmansard, 2016-07-25:
We need to add a unit test to validate the new integrate function against the jacobian methods, by computing an approximation of the jacobians by finite differencing. 

A preliminary test with jacobian COM indicates that there is a mistake somewhere.

jcarpent, 2016-07-26:
Can you give more details on such a mistake? 
Because jacobian of CoM is already checked against CRBA, which validated against RNEA, which is also validated against RBDL and so on ...

But anyway, it is still interesting to have the finite diff implemented inside the Pinocchio corpus.

nmansard, 2016-07-26:
Finitie differencing Jcom gives I3 in the 3 first columns, while it should be oR1.

I agree that the current Jcom algo is rock solid. I am rather afraid of a bug in the integrate. It is also possible that I am making a mistake with how to finite-diff on a Lie group.

I will push in the morning the corresponding unit test in hpp-pinocchio (you can already have a look in my branch topic/Jcom of hpp-pinocchio). We may want to add a similar test in pinocchio later.

jmirabel, 2016-07-26:
In hpp-constraints, class `DifferentiableFunction` has two algorithms for finite difference. It may help.

jcarpent, 2016-07-26:
After discussing with @jmirabel, it seems the bug comes from the joints with angulars transformation as FreeFlyer or SO3 joint. I will fixe it today.

nmansard, 2016-07-28:
See #278 . I close this issue as #278 is more generic and cover it.

closed 2016-07-28 by nmansard