stack-of-tasks/pinocchio #257: Change FCL data to Eigen

jcarpent, 2016-07-26:
This issue follows humanoid-path-planner/hpp-pinocchio/issues/10.

It might be useful to pass all the FCL linear algebra type to their eigen counterpart.

nmansard, 2016-07-26:
I suggest to keep this discuss for later depending on the choice made by the hpp team.
Any opposition to closing this issue and reopening it in September?

jcarpent, 2016-07-27:
It is already done in HPP, the change from internal linear algebra to Eigen version. So, from this remark, it is straightforward to move on.

jmirabel, 2016-07-27:
I agree with Nicolas for delaying the update in Pinocchio to September for two reasons:
- there are more serious concern before this (doing it or not does not affect us right now),
- @florent-lamiraux may have a good reason for one choice or another.

jcarpent, 2016-08-06:
Just to mention a few, FCL is moving to Eigen (flexible-collision-library/fcl/pull/150).

closed 2017-10-10 by jcarpent