stack-of-tasks/pinocchio #259: Use njoint instead of nbody for joint visitors

jcarpent, 2016-07-26:
Currently, nbody is used while njoint must be used inside all algorithms. This is a bug which must be solved quickly.

nmansard, 2016-07-26:
BTW: I dont like this redundancy of information, by explicitly storing the size of the lists (e.g. have njoint which is equal to joints.size()). Dont you think that we should get rid of it?

jcarpent, 2016-07-26:
Indeed. And it must be the same for nbody.

nmansard, 2016-07-26:
Ok. Should we remove nbody first and use njoint instead, and then do a second PR (later, less urgent) removing njoint, ngeom etc?

jcarpent, 2016-07-26:
I agree as njoint is safe.

jmirabel, 2016-07-26:
This means that, from now on, we should use `model.joints.size()` rather than `model.joints.njoints`, right ?

jcarpent, 2016-07-26:
Both `model.joints.size()`and `model.njoint` are ok. But to avoid redundant info, it might interesting to only consider the first solution. 
Right, you should avoid to use `model.nbody`which may lead to bugs as nbody >= njoint

nmansard, 2016-07-26:
For njoint: it rather was a question. Should we keep it or remove it. Maybe keep it is cheap and easy to assert, and simplify downstream use. 
In any case, we should mark nbody as deprecated.

jcarpent, 2016-08-02:
Solved by #288.

closed 2016-08-02 by jcarpent