stack-of-tasks/pinocchio #263: Compilation time is too long

nmansard, 2016-07-26:
We must work to reduce the compilation time, both in Pinocchio and for downstream libraries.

Some ideas:
1. move code from .hxx to .cpp and compile them => this is bad is later using a template version of the scalar type (yet hard coded double)
2. carefully work on what is included, to let the use explicitly choose what he wants to include => this is good in principle, but not so brilliant in the present case. I believe that most of the code is already properly separated (more or less by modules). We may try to only compile the needed joints. Any idea of how to do so? I dont believe it possible.
3. We could think of a macro to deactivate the code (ifndef) in hxx files. We then could keep the code in .hxx files that would not be loaded in downstream header files but only in downstream .cpp files. 
4. In addition, we could then also have trivial .cpp files that would simply include .hxx files for compilation in the pinocchio lib. In the perspective of having a header-only package (templated by scalar), we could pre-compile in pinocchio lib the scalar=double version of the code, that would not be re-compiled downstream, while letting the possibility to third party users to compile with another scalar type. 
5. As suggest in #247 , we may want to break the python module in several .cpp files, each being compiled in a simpler .o file, while letting the possibility to parallelize the compilation with make -j8.

jcarpent, 2016-08-04:
Partially solved by #286.

jmirabel, 2016-09-07:
Another ideas to reduce compilation time, inspired by Eigen design:

Split the hxx in separated files and invert the inclusion order (hxx includes hpp, and not the contrary). Let say a user needs only to access to the frames of a Model. He only needs to have the definition of functions related to the frames. He would include a file (say "pinocchio/multibody/frame") that would include "definition_of_frame_related_method_of_model.hpp" which includes "frame.hpp" and "model.hpp".

This means "model.hxx" is split in several files.

I don't know if it is possible because of the `variant`...

jcarpent, 2018-06-08:
The compilation have been improved for Python bindings.
Some remain works are still necessary, but they are no more a priority.

closed 2018-06-08 by jcarpent

jmirabel, 2020-04-16:
Bumping into the issue again. Now that pinocchio has grown, when I compile it, I see many things in the bindings which aren't interesting for me. Would it make sense to gather some of them into optional groups (when compiling from source only, not for binary packages) ?

For instance, I never use the derivatives.