stack-of-tasks/pinocchio #278: Check distance, interpolate and differentiate for each joint

jcarpent, 2016-07-28:
Currently, joint methods distance, interpolate and differentiate are full of bugs and the current unit tests are incorrect, and cannot highlight them.

nmansard, 2016-07-28:
Thanks for the report. 

@fvalenza could you make a brief status of how these methods are checked VS hpp-model? Are you planning to add such comparative test?

nmansard, 2016-07-29:
Currently, the unittest assert difference() against integrate(), which is a good test. However, both methods were wrong, so the test was misled.

@jcarpent modified integrate, and asserted it using se3 exponential map of the cartesian velocity resulting of qdot. This is also a good test.

Now integrate is valide (should be at least), but difference is not, so the initial test of @fvalenza is not accepted. @jcarpent deactivated it. 

Two steps are now necessary: 
1. fix difference, and possibly interpolate, and assert there results wrt integrate.
2. bullet-proof integrate by using it to compute (by finite diff) an approximation to the robot jacobians, and assert the value with respect to the (bullet proof) jacobian algorithms.

@jcarpent volunteered to implement a correct version of difference and interpolate.
@fvalenza would you agree to implement or re-implement the unit test checking difference and interpolate wrt to integrate.
Who can implement the final test checking integrate vs jacobian?

fvalenza, 2016-07-29:
Ok

jcarpent, 2016-08-04:
It seems to be solved by #292.

closed 2016-08-04 by jcarpent