stack-of-tasks/pinocchio #281: Refactoring Geometry

nmansard, 2016-07-28:
I would like to implement the following modification in multibody/geometry.hpp

1. Remove classes DistanceResults and CollisionResults and directly work with FCL native types. For distance results, possibly add the 3 member functions as static functions (although I do not think they are vital). Keep operator== outside of the classes.

2. Remove NONE from the enum.

3. Move all the definitions before struct GeometryModel in a geometry-fwd.hpp

4. move inner/outer objects from GeomModel to GeomData. Stop automatically feeling them at model construction time. Possibly fill them automatically at data construction time, or explicitly with a dedicated method.

5. rename all xxx_yyy as xxxYyy, except oMg_fcl

6. It was suggested by @stonneau to rather store in data a CollisionObject that a fcl::Transform, which makes the call to fcl computations easier. 

7. Remove ref to GeomModel from GeomData. Consequently change the signature of the algorithms using GeomData, to explicitly add GeomModel as an argument.

8. Remove methods computeAllCollisions and computeAllDistance, as they are redundant with the algorithm geometry.hpp

9. Possibly also remove computeCollision and computeDistance, and put the two as external functions in algorithm/geometry. Change their internal process to avoid (stack) memory allocation of a Collision/DistanceResult object.

10. Remove displayCollisionPair.

11. Add a PairIndex type (like JointIndex and GeomIndex) to make arguments of some methods more explicit.

Opinion, in particular from @fvalenza, welcome. Please help.

nmansard, 2016-07-29:
@jcarpent agreed on 6. I start implementing 6 that I need for hpp::pinocchio

fvalenza, 2016-07-29:
1/ If we work directly with fcl type, we lose the info concerning what are the collision pair ( the two GeomIndex) involved in this particular result.
2/ No particular opinion.  It was to force user to understand what is loaded by specifying it explicitly ( i.e using the GeometryModel as a CollisionModel or a VisualModel)
3/ Agreed. Some structs are even binded in python, so we could even put them in their own file may be?
4/ If we want to add/remove outerobjects or innerobjects on the fly, it may be better. Do we need to consider updated the list of collision pairs then ?
6/ As previsouly stated, it's the tradeoff of redundant information (the pointer on the collisiongeometry is both in GeometryModel and GeometryData then) vs easier calls to fcl computations. I personnally prefer storing the CollisionGeometry in the Model (as it is constant) and the transform in the Data. Just my personnal pov.
7/ Agreed, some GeomData methods work with the number of geometry objects. We need to propagate this information in GeomData if it' still needed after rework ( see point 7/8 particularly)
5/8/9/10/11/ OK

nmansard, 2016-07-29:
Thx.

1. We loose it directly, but we can easily recover it in the corresponding position of the collisionPairs vector, right?
geomData.collision_results[i] ==corresponds to==> geomModel.collisionPairs[i]

2. I think this is rather a design problem. By adding NONE, you check the choice of the user at run time (in DEBUG only, with an assert), while by not having it in the enum, the choice is checked at compil time, as it becomes a mandatory (not optional) argument of the parsing methods.

3. I dont think the naming is very important. I would tend to prefer to limit the number of header files, and name it "-fwd" makes it clear that it is of lesser importance. Maybe -fwd is not the good suffix though.

7. I would propose to have inner/outer lists as protected object, that can be accessed only with read-only method.

fvalenza, 2016-07-29:
As part of the #190 , i am doing point 3. I may PR only this point or point 3 and the issue together this afternoon.

nmansard, 2016-07-29:
Thanks for your help.

Then lets decide the name of the include. I think fwd is not so good, as it is not exactly forward declarations. 
I suggest fcl.hpp (ie pinocchio/multibody/fcl.hpp). It should contains all the fcl #include that we are needing in Pinocchio, and the wrapping, typedef and correspond #ifdef of the fcl classes.

nmansard, 2016-07-29:
I pas this issue from question to allocated. @fvalenza does 3 first (de-assign yourself when done). If somebody wants to do the rest, assign yourself, or assign me if nobody else volunteer.

nmansard, 2016-08-30:
I did not manage to work on that topic during August, so I remove the label.

jmirabel, 2016-09-05:
Pull-request #302 already provides the following changes:
- removes class `CollisionResult`,
- memory allocation in `computeCollision`.

Outer objects are not used in HPP (I have to double check this assertion). I think inner objects make sense in `GeometryModel`.

I agree with points 1, 2, 3, 5, 10, 11. I have no opinion for point 6, 8. 7 implies that the use of GeomData is slightly more complex.

nmansard, 2016-09-09:
solved by #314

closed 2016-09-09 by nmansard